hjd1964/OnStep #32: Move gear ratios from spreadsheet to config

kbahey, 2018-01-03:
In order to reduce confusion, we should let the machine derive the values rather than calculate them externally.

The worksheet will still be used for planning, and doing 'what if' scenarios on various gear ratios, microsteps, ...etc. and to ensure that the values chosen are sane, and not exceed the hard limits. 

So, the configuration Config.xxx.h can start with something like this:
```
// Number of steps per full rotation for your specific stepper motor.
// This should be 200 for 1.8 degree motors, or 400 for 0.9 degree motors
Axis1MotorStepsPerFullRotation     200

// Microsteps used for sidereal tracking. You can't use a lower fraction
// (i.e. higher number), unless your stepper driver supports it. For example,
// A4988 has a minimum of 1/16, DRV8825 has 1/32, LV8729 has 1/128.
// The Trinamic TMC2100, TMC2130, TMC2208 does interpolation of 1/256 internally
Axis1MicroStepsForSidereal         64

// This is any transfer gears or pulleys that are in between the worm wheel
// and the motor. Use 1 if you directly coupled the motor to the worm gear.
// Don't forget to account for the gears inside the motor if yours have them
Axis1TransferGearRatio             3

// This is your main worm wheel number of teeth. For CG5 and other Vixen Super
// Polaris clones, this is usually 144
Axis1WormWheelStepsPerFullRotation 144

// Same as above for your other axis
Axis2MotorStepsPerFullRotation     200
Axis2MicroStepsForSidereal         64
Axis2TransferGearRatio             3
Axis2WormWheelStepsPerFullRotation 144
```

Then in Globals.h, we can add the following snippet:

```
#define Axis1StepsPerWormRotation (Axis1MotorStepsPerFullRotation * Axis1MicroStepsForSidereal * Axis1TransferGearRatio)
#define Axis2StepsPerWormRotation (Axis1MotorStepsPerFullRotation * Axis2MicroStepsForSidereal * Axis2TransferGearRatio)

#define Axis1StepsPerDegree  ((Axis1StepsPerWormRotation * Axis1WormWheelStepsPerFullRotation) / 360)
#define Axis2StepsPerDegree  ((Axis2StepsPerWormRotation * Axis2WormWheelStepsPerFullRotation) / 360)
```

If you like this idea, I can create pull requests for Global.h and all the configs.

hjd1964, 2018-01-03:
I'm not liking this one.

kbahey, 2018-01-03:
The benefit here is that when someone asks a question (like Pete yesterday), you only need the config to troubleshoot it. No spreadsheets. You can use your own spreadsheets. You have the raw data right there to debug it. 

It also makes it obvious for those configuring stuff (again, Pete asked about gear ratios before we pointed him to the spreadsheet).

hjd1964, 2018-01-04:
Still don't like it.

- There is a simplicity to the way I'm doing it, straight to the heart of the matter.
- The StepsPerDegree can apply to servo based step/dir also.
- Gear train design isn't always two stages.
- More options in the config means more likely to get something wrong.
- Spreadsheet allows for gear train "what-if's"
- Users usually don't have an issue with this

Much more often they are confused about micro-step mode switching and programming modes.  But that's a difficult thing to make "simple", too many darn permutations.

kbahey, 2018-01-04:
The spreadsheet will still be there for what-if and such. Not replacing it. In fact, I was thinking of having a doc directory in OnStep github and committing the spreadsheet there, so people have less places to visit and less (virtual) parts to source.

But you are against it, so let us shelf that ...

Regarding microsteps and mode switching, see my email in the Pete MiniPCB thread. It addresses exactly this issue, and I suggested improvements there. I can commit them and update the wiki if you agree.

closed 2018-01-05 by hjd1964