hjd1964/OnStep #84: STM32 Tracking Precision - Observations and Discussion

kbahey, 2018-11-16:
I have been doing more testing on the tracking precision. I thought that here we can have a developer discussion without confusing the end users. 

I have 3 configurations:

- My GEM, 28,800 steps per degree
- Verloop's GEM, 1,777.777 steps per degree
- Zeppy's Alt-Az 12,214.793 steps per degree

First observation: increasing the seconds to average from 40 to 120 in Timer.ino:ClockSync() improves precision. 

Second, increasing 2L to 4L seems to make the 3 mounts happy, as below:

My GEM, 120 seconds, 4L 
2018-11-15 21:51:42 04:35:55#+16*30:29# 68.980174,16.508056#
2018-11-15 21:53:42 04:35:55#+16*30:29# 68.980243,16.508056#
2018-11-15 21:55:43 04:35:55#+16*30:29# 68.980264,16.508056#
2018-11-15 21:57:43 04:35:55#+16*30:29# 68.980354,16.508056#
2018-11-15 21:59:43 04:35:55#+16*30:29# 68.980403,16.508056#
2018-11-15 22:01:43 04:35:55#+16*30:29# 68.980465,16.508056#
2018-11-15 22:03:43 04:35:55#+16*30:29# 68.980528,16.508056#
2018-11-15 22:05:43 04:35:55#+16*30:29# 68.980590,16.508056#
((68.980174-68.980590)/(5+9))*60*60 = **0.106"/min**

Verloop's GEM, 120 seconds, 4L
2018-11-15 22:11:15 04:36:53#+16*33:30# 69.221271,16.558313#
2018-11-15 22:13:15 04:36:53#+16*33:30# 69.220875,16.558313#
2018-11-15 22:15:15 04:36:53#+16*33:30# 69.221000,16.558313#
2018-11-15 22:17:16 04:36:53#+16*33:30# 69.220896,16.558313#
2018-11-15 22:19:16 04:36:53#+16*33:30# 69.220834,16.558313#
2018-11-15 22:21:16 04:36:53#+16*33:30# 69.220771,16.558313#
2018-11-15 22:23:16 04:36:53#+16*33:30# 69.220709,16.558313#
2018-11-15 22:25:16 04:36:53#+16*33:30# 69.220084,16.558313#
2018-11-15 22:27:17 04:36:53#+16*33:30# 69.220334,16.558313#
2018-11-15 22:29:17 04:36:53#+16*33:30# 69.220229,16.558313#
2018-11-15 22:31:17 04:36:53#+16*33:30# 69.220084,16.558313#
2018-11-15 22:33:17 04:36:53#+16*33:30# 69.220021,16.558313#
2018-11-15 22:35:17 04:36:53#+16*33:30# 69.219438,16.558313#
((69.221271-69.219438)/(35-11))*60*60 = **0.204" to 0.305"/min**

Note that his goto pointing is a bit off.

Zeppy's Alt-Az, 120 seconds, 4L
2018-11-15 22:40:51 04:36:56#+16*33:23# 69.233334,16.556401#
2018-11-15 22:42:51 04:36:56#+16*33:23# 69.233299,16.556461#
2018-11-15 22:44:51 04:36:56#+16*33:24# 69.233306,16.556552#
2018-11-15 22:46:51 04:36:56#+16*33:24# 69.233357,16.556624#
2018-11-15 22:48:51 04:36:56#+16*33:24# 69.233313,16.556761#
2018-11-15 22:50:52 04:36:56#+16*33:24# 69.233272,16.556850#
((69.233334-69.233357)/(46-40))*60*60 = **0.022"/min in RA**
There is something else going on in DEC precision. I am not sure why that is. 
Also, Goto is a bit off.

First, I hope my calculations are correct. 
The results are certainly better than 2L, and 40 seconds in ClockSync.

I played around with increasing the value of 4L more, and even adaptive code (e.g. if StepsPerDegree for axis >= 28000 use 32L or 1024L, if StepsPerDegree for axis > 12000 use 4L, ...etc.) but could not conclude something concrete that will work for all configurations.

I also thought that there may be a Config.STM32 parameter that could override the 4L
When you get a chance of building the STM32, the new :GXFE# command may come in handy, and you can validate the results above. I am sure you will also have more insight into the underlying reasons for all this, rather than my trial and error method so far.

In the meantime, it seems that 4L is a good compromise providing better tracking across the board.

hjd1964, 2018-11-16:
Guessing you mean changing the from 2L to 4L in timer3SetInterval, etc.  I really prefer to not get dragged too far into this STM32 stuff but I have to say I don't understand the reason behind the changes.

That 2L/4L (etc. for other platforms) are there to constrain the required counts to the timer counter range with the overflow going into the auxiliary software counter.  Since the STM32 calls being used automatically set the pre-scaler the max "timer counter range" should be 2^32 (with a bit of resolution sacrificed every 2^16 as the count increases.)  You're way under the 134 maximum period I impose, this isn't a big deal really since the step rates are so low most would never notice but it's there for Alt/Az mode where in some areas of the sky Alt or Az will barely move.

It's difficult to say what's going on but it's almost like the values you're playing around with are making you happy since they're shifting resolution up into the software counter so you don't trigger the pre-scaler change in you test cases...

Now on the Mega2560 (and this was ages ago when I wasn't as capable a C/C++/Arduino coder as I am now) I gave up on setting the pre-scaler on-the-fly in the motor timers due to the timing irregularities it caused.  Any extra code in those motor timer ISR's to figure out the pre-scaler is bad due to more ISR contention/timing jitter, slower ISR's.  Then there's a hidden counter in that pre-scaler when you re-program it do you end up taking one tick extra fast or extra slow before if falls into the new cycle?  And on the STM32 the whole idea of doing that refresh() thing makes me wonder.  Are we forcing a counter reset?  That's not good, you don't want the counter to stop even as the ISRs' are being executed that will cause rate problems.

hjd1964, 2018-11-16:
As for PPS I guess there's more jitter in the measurements on STM32 vs. other platforms.  I'm not even sure what priority level the pin interrupts run at on the Teensy but measurements show there's no significant accuracy problem.  On the Mega2560 there are no interrupt priority levels but the biggest slowest sidereal clock ISR runs with "no block" so only the motor ISR's block that pin interrupt.  The ESP32 also doesn't seem to have any major issues (not as good as the Teensy but better than STM32 at about 0.01 arc-sec/minute in Eq mode.)  Again, keep in mind this isn't a direct measure of step rate accuracy but rather of agreement between the sidereal clock and step rate.

kbahey, 2018-11-17:
Thanks for the insights.

Today, I am running a long duration test with the mount attached to see the full behavior.

It is with 4L (instead of 2L in the timer[34]SetInterval() function, and 120 seconds averaging.
This uses my configuration, a GEM with 28,800 steps/deg
Star is Mirfak, in Perseus

3 hours 10 minutes, and 3.95 arc seconds of drift. So, 0.065 arc sec/min. Acceptable for my purposes.

(the last two columns are calculations on the fly by subtracting max RA from min RA and dividing by the minutes, so arc seconds per minute)

2018-11-16 17:58:22 03:25:28#+49*56:21# 51.366146,49.939201  0.000  0.000
2018-11-16 17:59:22 03:25:28#+49*56:21# 51.366181,49.939201  0.126  0.000
2018-11-16 18:00:23 03:25:28#+49*56:21# 51.366181,49.939201  0.063  0.000
2018-11-16 18:01:23 03:25:28#+49*56:21# 51.366222,49.939201  0.091  0.000
2018-11-16 18:02:23 03:25:28#+49*56:21# 51.366257,49.939201  0.100  0.000
2018-11-16 18:03:23 03:25:28#+49*56:21# 51.366285,49.939201  0.100  0.000
2018-11-16 18:04:23 03:25:28#+49*56:21# 51.366285,49.939201  0.083  0.000
...
2018-11-16 19:04:17 03:25:28#+49*56:21# 51.367340,49.939201  0.000  0.000
2018-11-16 19:05:18 03:25:28#+49*56:21# 51.367340,49.939201  0.000  0.000
2018-11-16 19:06:18 03:25:28#+49*56:21# 51.367375,49.939201  0.063  0.000
2018-11-16 19:07:18 03:25:28#+49*56:21# 51.367403,49.939201  0.076  0.000
2018-11-16 19:08:18 03:25:28#+49*56:21# 51.367417,49.939201  0.069  0.000
2018-11-16 19:09:18 03:25:28#+49*56:21# 51.367417,49.939201  0.055  0.000
...
2018-11-16 20:05:17 03:25:28#+49*56:21# 51.368451,49.939201  0.000  0.000
2018-11-16 20:06:17 03:25:28#+49*56:21# 51.368486,49.939201  0.126  0.000
2018-11-16 20:07:18 03:25:28#+49*56:21# 51.368521,49.939201  0.126  0.000
2018-11-16 20:08:18 03:25:28#+49*56:21# 51.368521,49.939201  0.084  0.000
2018-11-16 20:09:18 03:25:28#+49*56:21# 51.368507,49.939201  0.063  0.000
2018-11-16 20:10:18 03:25:28#+49*56:21# 51.368549,49.939201  0.071  0.000
...
2018-11-16 21:05:05 03:25:29#+49*56:21# 51.369556,49.939201  0.000  0.000
2018-11-16 21:06:05 03:25:29#+49*56:21# 51.369556,49.939201  0.000  0.000
2018-11-16 21:07:06 03:25:29#+49*56:21# 51.369590,49.939201  0.061  0.000
2018-11-16 21:08:06 03:25:29#+49*56:21# 51.369625,49.939201  0.083  0.000
2018-11-16 21:09:06 03:25:29#+49*56:21# 51.369625,49.939201  0.062  0.000
2018-11-16 21:10:06 03:25:29#+49*56:21# 51.369618,49.939201  0.050  0.000

Regarding the PPS, we can bury the value as a define in HAL. So on STM32 it is 120, and on others, it is 20 as before. That way, no impact on other platform, no regressions, ...etc. This is easy to do, and I can submit a pull request if you want.

Regarding STM32's timer, we are explicitly calling pause(), which pauses the timer for microseconds.

This is the high level class [HardwareTimer](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/HardwareTimer.cpp), and it calls functions in this [low level file](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/system/libmaple/include/libmaple/timer.h).

If you look at line 105 in the first file, this is setPeriod(). Internally it sets the prescaler and overflow. But I assume that is what you want it to do. 

timer_generate_update() which underpins refresh(), does not say anything about resetting counters.

I have an idea: There is a getCount() and setCount() functions for the timers. Perhaps before pausing the timer, I should call the former, then call the latter with the saved value, after doing a refresh() followed by a resume(). This means the counts are not lost. 

One caveat is that the documentation says: "If this value exceeds the timer’s overflow value, it is truncated to the overflow value. "

Do you think this is worth a shot?

kbahey, 2018-11-17:
Well, that change of saving the timer count, then using it to prime it again did not work. Made things worse.

Maybe it is truncation, since setCount and getCount use uint16.

0.2"/min, worse than with 4L and 120 seconds

2018-11-16 21:32:54 02:07:12#+23*27:42# 31.798514,23.461701  0.000  0.000
2018-11-16 21:33:54 02:07:12#+23*27:42# 31.798479,23.461701  0.126  0.000
2018-11-16 21:34:54 02:07:12#+23*27:42# 31.798403,23.461701  0.200  0.000
2018-11-16 21:35:54 02:07:12#+23*27:42# 31.798368,23.461701  0.175  0.000
2018-11-16 21:36:54 02:07:12#+23*27:42# 31.798299,23.461701  0.193  0.000
2018-11-16 21:37:54 02:07:12#+23*27:42# 31.798222,23.461701  0.210  0.000
2018-11-16 21:38:54 02:07:12#+23*27:42# 31.798194,23.461701  0.192  0.000
2018-11-16 21:39:55 02:07:12#+23*27:42# 31.798125,23.461701  0.199  0.000

kbahey, 2018-11-17:
Okay, I have tested 4L for the 3 configurations, and created a pull request for it.

Test results in the pull request.

hjd1964, 2018-11-18:
I tested a STM32 a little bit here...

1. Using the default config file it didn't work for me at all.  The sidereal clock wouldn't start.  I changed the code for Timer1SetInterval() and removed the "firstPass" stuff and it started working.  Lol, I have no idea why this is working for everyone. :)

2. Slews into the eastern sky (west of the pier) ramp up speed then the RA step output goes crazy.  The STM32 doesn't crash but something very strange happens.  **I have no fix for this.**

3. In messing around with the STM32 HAL I noticed that TIMER_CHn don't seem to match up with the ea. correct channel.  I fixed that here but will leave it to you to correct/test and push the update.

4. QuickSetIntervalAxis2() with just setPeriod(r); works fine.  We don't really want to -1 since these I notice these functions (or how the timer is setup probably) seems to handle that aspect.  Often direct writing into the timer compare register needs one less than the period but again not here from what I can see.  

4A. Also, we do it twice (!) since in timernSetInterval() has it too!  This amounts to 2uS error in the setting (fast.)

hjd1964, 2018-11-18:
Some measurements:

Teensy3.5 (WITH PPS)
101.7636		<-Should be (Hz)
2E-05		20ppm
0.00204		20ppm error
101.7667		<-Measures (0.0031 Hz fast)

STM32 (NO PPS OR WITH PPS, 2L)
53.4794		<-Should be (Hz)
2E-05		20ppm
0.00107		20ppm error
53.5242		<-Measures (0.0448 Hz fast)

STM32 (WITH PPS, 2L, -1 fixes)
53.4794		<-Should be (Hz)
2E-05		20ppm
0.00107		20ppm error
53.4898		<-Measures (0.014 Hz fast)

STM32 (WITH PPS, 32L, -1 fixes)
53.4794		<-Should be (Hz)
2E-05		20ppm
0.00107		20ppm error
53.4845		<-Measures (0.005 Hz fast)

Above were done with a Teensy3.2 + OLED.  I plan to add GPS PPS for real accuracy (above based on it's 20ppm xtal only) and put it in a case at some point but good enough for now.

[image]

I emailed over the changes so you can see them and verify they work on other configurations.

hjd1964, 2018-11-18:
Note too: I still get that weird timer glitch problem, so my STM32 doesn't really work!  Before or after this change.

kbahey, 2018-11-18:
What timer glitch?

kbahey, 2018-11-18:
Here are the test results from the first set of changes you made (the Timer and HAL files you emailed me:

For 28,800, they are more than acceptable. 
`
Howard's changes 2018-Nov-18, 40 second PPS
GEM, Khalid, 28,800 Steps/Degree
2018-11-18 13:21:36 23:04:45 +15*12:18 346.187194,15.205035  0.000  0.000
2018-11-18 13:22:36 23:04:45 +15*12:18 346.187201,15.205035  0.025  0.000
2018-11-18 13:23:36 23:04:45 +15*12:18 346.187167,15.205035  0.061  0.000
2018-11-18 13:24:36 23:04:45 +15*12:18 346.187174,15.205035  0.041  0.000
2018-11-18 13:25:37 23:04:45 +15*12:18 346.187139,15.205035  0.056  0.000
2018-11-18 13:26:37 23:04:45 +15*12:18 346.187139,15.205035  0.044  0.000
2018-11-18 13:27:37 23:04:45 +15*12:18 346.187104,15.205035  0.058  0.000
2018-11-18 13:28:37 23:04:45 +15*12:18 346.187111,15.205035  0.050  0.000
2018-11-18 13:29:37 23:04:45 +15*12:18 346.187118,15.205035  0.044  0.000
2018-11-18 13:30:38 23:04:45 +15*12:18 346.187076,15.205035  0.050  0.000

For low spd, they are so so. 

Howard's changes 2018-Nov-18, 40 second PPS
GEM, Verloop, 1,777.777 Steps/Degree
2018-11-18 13:37:39 21:44:56 +09*59:14 326.232958,9.987188  0.000  0.000
2018-11-18 13:38:39 21:44:56 +09*59:14 326.233167,9.987188  0.752  0.000
2018-11-18 13:39:39 21:44:56 +09*59:14 326.232854,9.987188  0.563  0.000
2018-11-18 13:40:39 21:44:56 +09*59:14 326.232854,9.987188  0.375  0.000
2018-11-18 13:41:40 21:44:56 +09*59:14 326.232854,9.987188  0.281  0.000
2018-11-18 13:42:40 21:44:56 +09*59:14 326.233063,9.987188  0.225  0.000
2018-11-18 13:43:40 21:44:56 +09*59:14 326.233063,9.987188  0.187  0.000
2018-11-18 13:44:40 21:44:56 +09*59:14 326.233313,9.987188  0.235  0.000
2018-11-18 13:45:40 21:44:56 +09*59:14 326.232833,9.987188  0.215  0.000
2018-11-18 13:46:41 21:44:56 +09*59:14 326.233042,9.987188  0.191  0.000

Howard's changes 2018-Nov-18, 40 second PPS
AltAz, Zeppy, 12214.793 Steps/Degree
2018-11-18 13:49:53 23:04:51 +15*12:18 346.211012,15.205045  0.000  0.000
2018-11-18 13:50:53 23:04:51 +15*12:18 346.210905,15.205082  0.385  0.133
2018-11-18 13:51:53 23:04:51 +15*12:19 346.210874,15.205200  0.248  0.279
2018-11-18 13:52:54 23:04:51 +15*12:19 346.210753,15.205225  0.311  0.216
2018-11-18 13:53:54 23:04:51 +15*12:19 346.210697,15.205280  0.283  0.211
2018-11-18 13:54:54 23:04:51 +15*12:19 346.210685,15.205313  0.234  0.192
2018-11-18 13:55:54 23:04:51 +15*12:19 346.210663,15.205378  0.209  0.199
2018-11-18 13:56:54 23:04:51 +15*12:19 346.210634,15.205369  0.194  0.171
2018-11-18 13:57:55 23:04:51 +15*12:20 346.210527,15.205459  0.218  0.186
2018-11-18 13:58:55 23:04:51 +15*12:20 346.210498,15.205482  0.205  0.174

And just for the heck of it, I disabled PPS, and although the results are similar to the previous test, they are smoother.

Howard's changes 2018-Nov-18, PPS OFF
AltAz, Zeppy, 12214.793 Steps/Degree
2018-11-18 14:23:48 23:04:51 +15*12:18 346.211099,15.205164  0.000  0.000
2018-11-18 14:24:48 23:04:51 +15*12:18 346.211054,15.205189  0.162  0.090
2018-11-18 14:25:48 23:04:51 +15*12:19 346.210888,15.205238  0.379  0.133
2018-11-18 14:26:48 23:04:51 +15*12:19 346.210857,15.205271  0.290  0.128
2018-11-18 14:27:49 23:04:51 +15*12:19 346.210783,15.205341  0.284  0.159
2018-11-18 14:28:49 23:04:51 +15*12:19 346.210772,15.205398  0.234  0.168
2018-11-18 14:29:49 23:04:51 +15*12:20 346.210626,15.205442  0.283  0.166
2018-11-18 14:30:49 23:04:51 +15*12:20 346.210598,15.205540  0.257  0.193
2018-11-18 14:31:49 23:04:51 +15*12:20 346.210502,15.205579  0.268  0.186
2018-11-18 14:32:50 23:04:51 +15*12:20 346.210475,15.205621  0.249  0.182
`

Waiting for the second set of changes.

hjd1964, 2018-11-18:
The actual HA (hour angle) and so RA position is based on the counts and each count is > 2 arc-seconds for Verloop.  At about 12000 it's 0.3" at 24000 0.15" so much less noticeable around there.

Another potential source of jitter/error is the sidereal clock.  Probably should switch to a fixed pre-scaler for it too.  If PPS is off the sidereal clock timer doesn't get re-programmed during tracking since the rate doesn't change except before/after a goto.  Once set it stays as is during your test (for good or bad) but if PPS is ON the sidereal clock period is being adjusted now and then and if there's a timer design problem (pre-scaler/counter reset/etc.) it'll cause issues.

hjd1964, 2018-11-18:
Before I merge... 

Again, I can't do gotos and have no way to test if this really works except for tracking.  You can slew around the sky in kstars and everything seems to be working normally for your STM32?

kbahey, 2018-11-18:
What I am doing is slewing from some OnStep Python framework that I am working on. And when do GD and GR, I get the correct RA and DE (for my spds, a bit off for the other configs, as I noted in the results).

But yes, when I connect KStars, I see the pointer going to where I want when I slew (e.g. Alpheratz in Pegasus), and staying there when it is tracking.

Are you sure you are monitoring the right pin? When I used my el-cheapo oscilloscope, I would see square waves when slewing but not when tracking (maybe a limit to the resolution?)

hjd1964, 2018-11-18:
Design is as follows...

This new STM32 code is similar to the Mega2560 code.  The motor timers use a fixed pre-scaler that gives 0.25us per count.  There are two ISR calls to the step for the STM32 code path vs. 1 ISR call per step for the Mega2560.  So granularity is at the 0.5us level for each.  65536 (16 bit overflow register depth) * 0.5us per count (32768) gives a maximum period of 0.032768 seconds (about 30.5 steps per second.)  Longer than that and the overflow goes into the auxiliary counter that can range up to 4096 x as long (134 seconds.)  As that auxillary counter takes up the counts the granularity is increased (lower timing resolution... the ability to hit an exact rate.)  So best performance is >30.5 steps per second at 0.5us granularity.  Down to 15.25 steps per second 1us granularity.  Down to 7.625 steps per second 2us granularity. etc.

hjd1964, 2018-11-18:
Yes 100% sure the right pin... and no slewing in Sky Planetarium confirms it.  Down right broken.  Wonder if it's a compiler tool chain problem.

You should get a square wave always on STM32.

kbahey, 2018-11-18:
Definitely tracking and slewing flawlessly.

Here is the output from my Python framework.

First it was on Alpheratz in Pegasus
Tracking RA: 00:09:22 DE: +29*11:44
Tracking RA: 00:09:22 DE: +29*11:44

Told it to slew to Almach in Andromeda
Slewing RA: 00:24:51 DE: +32*49:44
Slewing RA: 01:03:22 DE: +41*18:30
Slewing RA: 01:42:09 DE: +42*26:06
Slewing RA: 02:04:45 DE: +42*26:06

Reached it and tracking ...
Tracking RA: 02:04:51 DE: +42*26:06
Tracking RA: 02:04:51 DE: +42*26:06
Tracking RA: 02:04:51 DE: +42*26:06

No issues. 

Confirmed from the app too.

Using 28,800 for SPD on both axes, but confirmed with the other two configurations as well.

hjd1964, 2018-11-18:
It's merged.  

I should note the pre-scaler setup is dependent on STM32 running at 72MHz.  We'd need to add code to make it work at any F_CPU.

kbahey, 2018-11-19:
> So best performance is >30.5 steps per second at 0.5us granularity. Down to 15.25 steps per second 1us granularity. Down to 7.625 steps per second 2us granularity. etc.

Great, I updated the [STM32 Wiki page](https://groups.io/g/onstep/wiki/STM32F103-Blue-Pill) to include this information.

Also, I made a [video showing the mount slewing and tracking.](https://www.youtube.com/watch?v=tgzy1bFrGBc) All works as designed. Thanks for the work, today and since 2013!

closed 2018-11-19 by kbahey