hjd1964/OnStep #91: Discussion on proposed SHC features and changes

kbahey, 2018-11-30:
Now that Dave Schwartz and myself have working SHC (based on ESP32), it is time to discuss new features and changes, to get consensus before submitting pull requests for them. 

- I like to move the Tracking menu from Settings to be under the main menu. This is because sometimes you need to start tracking to do certain tasks (e.g. polar alignment in KStars). It would be below Parking and above PEC.

- Related: I like to add the King Tracking rate, which is missing from the Android App (and KStars too).

- Related: is there a way to display the current tracking rate?

- Local time display is odd. Right now it is 16:12 EST, but the SHC displays 04:12. The UTC offset is 05 hrs. All this does not seem to affect how OnStep is running, since the STM32 has an RTC that stores the date/time/..etc.

- There is a discrepancy between the target RA/Dec, and what is slewed to. For example, I align on Alpha Andromeda and the SHC says are RA 00:08:12 and Dec +29:04:00. After the slew is complete, the display shows 00:09:03 and +29:00:00. This is verified by CdC over USB showing that it is off target. It seems that the calculation of the coordinates in CatMgr::ra() and CatMgr::dec() is not right. But before I tackle this, I am wondering if this is issue shows only on ESP32 SHC, or you are seeing the same on a Teensy SHC (which neither one of us has)

hjd1964, 2018-11-30:
Move tracking menu, ok.
King Rate I omit since refraction compensated tracking is superior.
A way to display tracking rate?  ok.
No idea about time display, investigate.
I did testing of gotos and found systematic errors in the catalogs (Charles corrected.)  After that fix a sampling of gotos to various objects was within the expected accuracy.  Your result is too far from correct to be due to the low accuracy catalogs so not sure what's going on.  I will double check to see if the problem exists on Teensy.

kbahey, 2018-11-30:
While you are at it, please check the Teensy for the time display too, to confirm or deny it is a common issue, or ESP32 only.

hjd1964, 2018-11-30:
A major fail in the date/time would throw off precession and thus the coords of objects.

I'd have to review the code to make a final determination on the accuracy... it's not great in the catalogs, not designed to use the full bit-depth of the data types.  Before vs. after coords being different probably due to J2000 vs. Jnow (refraction, etc.)  that is J2000 vs. Observed.

kbahey, 2018-12-02:
#92 Now has Tracking under the Main Menu.

Working on some of the rest.

kbahey, 2018-12-02:
The declination goto problem is that they are always nn:00:00 no matter what it is. 

I traced that to the Ephemeris library, using a sample sketch:

  float Dec = 40.35;
  int ivd1, ivd2;
  float fvd3;
  Ephemeris::floatingDegreesToDegreesMinutesSeconds(abs(Dec), &ivd1, &ivd2, &fvd3);
  Serial.print("Convert float to Dec:");
  Serial.print(ivd1);
  Serial.print(":");
  Serial.print(ivd2);
  Serial.print(":");
  Serial.println(fvd3);

Output is: 

Convert float to Dec: 40:0:0.00

I found a better way to get the minutes, using modf()

  float Dec = 40.35;
  int ivd1, ivd2;
  float fvd3;
 
  double frac, integer;
  frac = modf(Dec, &integer);

  ivd1 = (unsigned int)Dec;
  ivd2 = frac*60;
  fvd3 = Dec*3600 - ivd1*3600 - ivd2*60;

  Serial.print(ivd1);
  Serial.print(":");
  Serial.print(ivd2);
  Serial.print(":");
  Serial.println(fvd3);
            
Output is:

40:20:60.00

But it does not work for the seconds.

kbahey, 2018-12-02:
This snippet works standalone, in IDEOne too.
```
  float Dec = 40.36;

  int d   = (int)Dec;
  int m   = (int)((Dec - (float)d) * 60.f);
  float s = (float)(Dec - (float)d- (float)m/60.f) * 60.f * 60.f;

  Serial.print(d);
  Serial.print(":");
  Serial.print(m);
  Serial.print(":");
  Serial.println(s);
```

Output on serial monitor is like this:

40:21:36.00

Which is what we want. 

But when I set *minute = m and *second = s in the ephemeris function, zero is still returned due to pointers and what not ...

hjd1964, 2018-12-02:
There appears to be no such problem with Dec on the Teensy3.2 SHC.
Local date/time also displays correct on my SHC.

Code is roughly the approach I use.  I tested my routines that do the same by converting & comparing in both directions (numeric<-->text) and across a wide range of values and smaller ranges but at high resolution especially around where seconds tick to minutes and minutes tick to degrees.  Be sure to also +/- values too.

I've seen this type of code almost but not quite work too many times.

kbahey, 2018-12-03:
The Dec problem is that when converting the decimal value to DD:MM:SS, only the degrees show up and minute and second are always zero. 

I traced it using DL()s, and it is caused by the abs() call. Fixed it and tested it north and south of the celestial equator. 

Pull Request #92 has a fix for this now.

hjd1964, 2018-12-03:
I see some other places where this would break code in the SHC (and OnStep too.)
I assume using fabs() would fix those.

hjd1964, 2018-12-03:
Looks like they use a macro on the ESP32 though...
#define abs(x) ((x)>0?(x):-(x))
Why the heck would that not work?

kbahey, 2018-12-03:
No idea, but my modification is tested and verified to work.
 
Also the call to the method with an abs() is not needed. Ephemeris::floatingDegreesToDegreesMinutesSeconds(abs(Dec), &ivd1, &ivd2, &fvd3);
Because inside that method, there is a check for negative and positive values for the float degrees. 

I will work on the date issue. After that, I will work on more precise coordinates for the objects, which I already extracted from the HYG database (stars, Messier, but not Herschel 400). 

There is also the Caldwell catalog of 110 nice fairly bright objects.

hjd1964, 2018-12-03:
"Also the call to the method with an abs() is not needed. "

I have no objection to the approach Charles used there and unless you want to re-code the other routines called it's probably best to not worry about it.  At-least I know where the sign is carried through at a glance.

For catalogs I'd have scaled so RA 0..24 is 0..65536 and Dec -90..90 to/from -32767..32767 for about 20 arc-seconds in RA and 10 arc-seconds in Dec.

kbahey, 2018-12-03:
I am big fan of the "if it works, don't fix it" motto ... 

No need to recode anything else re: abs(). It was just that single call to the Ephemeris  method with an embedded abs(), which is not needed at all, since the method handles the sign internally. Doing the abs() afterwards works the same.
So, what is in Alpha now is good, and works. 

For the coordinates in the catalogs, I find there is an offset when an star is slewed to. Varies from 4'52" to 2' mostly westwards, which points to an epoch, or equinox problem. For Messier objects, the error is not more than 2' from the few I tested. For Herschel, it is as big as for the stars. 

What I really would like to do for the catalogs is make them more maintainable. We have scripts that convert a CSV file (or a SQL database) to C structures, so we don't have to edit anything manually. I already have a list of stars brighter than magnitude 4, extracted from the [HYG database](https://github.com/astronexus/HYG-Database), where everything is of epoch and equinox 2000. I also have a Messier list, Caldwell (110 objects). Working on an NGC one but that is not a priority.

One issue is that Charles structures are fragmented into several entities (one for star RA, another for star DEC, a third for magnitude, and so on). Ideally, the ultimate goal would be to have single array structure per catalog. I don't know how hard is this to do. Perhaps I will open a separate issue for it for tracking it as a task to be done whenever someone gets to it.

hjd1964, 2018-12-03:
I'd use something like this...

typedef struct CelestialObj
{
const short Dec;
const unsigned short RA;
// etc.
} CelestialObj;

CelestialObj Stars[2] = {{1,1},{1,1}};

hjd1964, 2018-12-03:
Remember keep the data as small as possible, these are big catalogs for a Teensy3.2, 256K of flash gets used up fast.

Also use "const" to be sure the data doesn't end up in "memory" but in flash.

kbahey, 2018-12-03:
OK, noted. 

Will try to prototype this, and see if the slews are more accurate. If they are, then the effort is worth it, and I will redo all the catalogs.

kbahey, 2018-12-04:
#95 has a fix for the display of local time. 

The other issue is the peculiar way the SHC uses for setting the time. When going to Menu -> Settings -> Date/Time, 00:xx:xx means 12:xx:xx pm. Just takes getting used to.

kbahey, 2018-12-04:
All the issues reported here are now addressed.

The only remaining one is a task: better accuracy for the catalogs, and more catalogs (e.g. Caldwell). I am working on those.

redorbit, 2018-12-04:
One thing which is much different you wrote guys but can be disturbing.

If slewing to target and holding down the appropriate direction button on SHC
Once you got to the target you release the button but the controller drives forth a lot due to gradual deceleration.
So than you have to drive back manually in opposite direction and sometimes you have to repeat this through more cycles until you find back to desired target.

So if it would be not a big project I would prefer to get back automatically to the position where button was released.
Thanks

Peter

Closed #91.

hjd1964, 2018-12-04:
Peter is talking about deceleration during a guide operation.  His proposed solution isn't logically workable as it would cause way more confusion than it would solve.  

I suggest using a slower guide rate and/or tapping the guide button to control the slew rate as you approach where you expect the object to be so you don't overshoot by much.

redorbit, 2018-12-04:
> @redorbit I have not seen this issue myself.
>
> Why would you hold a button during slewing at all?
>
> To start a slew, one presses the 'Right' button and releases it.
> If any button is pressed during a slew, that causes the slew to aborts.
>
> This issues is closed, so please open a new one and describe the feature request you need in more detail (actions taken, expected results, vs. actual results).

I mean if I don’t want to goto to a specific target but find “manually” for example to find a star which is bright enough to guide my telescope during exposures.
I select a speed of 16x or 32x and scroll on the sky...

OK i’ll open a new one for sure ;)
Thanks
Peter

hjd1964, 2018-12-04:
Peter,
Are you using a Mega2560 based OnStep?

redorbit, 2018-12-05:
Hi Howard
No, currently a MiniPCB with Teensy 3.2

hjd1964, 2018-12-05:
The Mega2560 can be a little sluggish responding to SHC guide button presses.  The Teensy's I'd say are perhaps not as responsive as a basic hand controller (simple switch closure type) but it's quick enough to not annoy me.  At 16x or 32x speed the deceleration thing doesn't usually take much time I can't hardly tell at those speeds (but this still depends on your configuration.)  

At half-max or max guide speed deceleration must be planned for but that's not abnormal.  You can't just decelerate more quickly than a goto or you run the chance of stalling the stepper motors.

hjd1964, 2018-12-05:
More for Khalid/Dave...
I plan to add commands to the ST4Serial class to pass guide commands as one byte control codes.  There's plenty of ASCII that LX200 doesn't ever use so I will pick 8 chars, 4 for starting NSEW (similar in function to :Me# for example) and 4 for ending NSEW (:Qe#.)  These can then be embedded right into the middle of a LX200 conversation no waiting for the next command to finish on the interface.  At OnStep the command is intercepted and processed without being inserted into the usual general purpose LX200 command processing.  This is why I wait to enable the other "true" ST4 interface on the SHC.

redorbit, 2018-12-05:
Yes it’s a problem rather at higher speeds
So I can live with it
Thanks
Peter

kbahey, 2019-03-05:
Here are some suggested improvements and changes to the SHC.

1. For stars, the seconds part of the DEC is not shown (always 00), but that does not affect accuracy. Need to be fixed for user interface reasons.
2. There are more object types than the icons for them in the u8 code. The latter has only 5: Planetary, Emission, Galaxy, Globular and Open. The other types (e.g. Hii region, Reflection Nebula, SNR, ...etc.) can be added. But I am not the one to create icons and such (don't do graphics). 
3. For Messier, we can add the object type as a text, since there is room for that.
4. For stars, instead of using the Greek letter for the Bayer designation, we can use the 3 letter Latin equivalent (e.g. Alp, Bet, ...etc) since many people do not know the Greek alphabet. We can also combine this Latin abbreviation with the full constellation name (so Alp Andromeda, Bet Auriga, and so on). There is enough room on the screen.
5. We also have the magnitude of the star stored, so we should display it.
6. I did not add the common name for the stars, since not all of the 400 stars have it, and it will require about 10 bytes per star, so about 4,080 more bytes. If we add the full constellation name (as above), there will not be room on the screen for the common name for the star.

kbahey, 2019-03-05:
All of the above points have been addressed, except for one.

That is the Greek vs. Abbreviation. It can be added if we remove the labels for RA and DE, and replace them by the 3 letter abbreviation for the Greek letter and the constellation, respectively. Then the Common name and magnitudewill be on the first line. 

But that means the "Alp And" will be broken on two lines, which may confuse some people.

So I will not do this until we have consensus ...

closed 2019-03-05 by kbahey