hjd1964/OnStep #96: Overhaul the Catalogs with more accurate data, and new data

kbahey, 2018-12-04:
Progress on extracting star and DSO data.

First the data structure to store this in (goes into Catalog.cpp).

```
typedef star {
  const char Constellation;
  const char Bayer;
  const short Magnitude;
  const short RA;
  const short DE;
} star;

star cat_star[211]  = {
// Stuff generated from script below go here
};
```

A similar structure for DSOs would add object_type, and ditch the Bayer designation.

Now, extracted data from the [HYG Database](https://www.astronexus.com/hyg). 

For stars, the extract/transform program requires the magnitude as a command line  argument (star fainter than this will not be extracted):

The fields are: 

- Constellation (need to be converted to a char)
- Bayer Designation (need to be converted to a char, maps to the Greek letter)
- Magnitude (multiplied by 1,000)
- RA (multiplied by 10000)
- Declination (multiplied by 10,000)

The comments on the same line preserve the original values so as to make debugging easier, as well include the proper name

Here is an example run, which returns 211 stars
```
./extract-stars.py 3.2 | sort | less
{And, Alp, 2070, 1398, 290904}, // And Alp Alpheratz 2.070 0.139791 29.090432
{And, Bet, 2070, 11622, 356206}, // And Bet Mirach 2.070 1.162194 35.620558
{And, Gam-1, 2100, 20650, 423297}, // And Gam-1 Almaak 2.100 2.064984 42.329725
{Aql, Alp, 760, 198464, 88683}, // Aql Alp Altair 0.760 19.846388 8.868322
{Aql, Gam, 2720, 197710, 106133}, // Aql Gam Tarazed 2.720 19.770994 10.613261
{Aql, Zet, 2990, 190902, 138635}, // Aql Zet  2.990 19.090169 13.863478
...
{UMi, Alp, 1970, 25298, 892641}, // UMi Alp Polaris 1.970 2.529750 89.264109
{UMi, Bet, 2070, 148451, 741555}, // UMi Bet Kochab 2.070 14.845105 74.155505
{UMi, Gam, 3000, 153455, 718340}, // UMi Gam  3.000 15.345483 71.834016
{Vel, Del, 1930, 87451, -547088}, // Vel Del  1.930 8.745059 -54.708821
{Vel, Gam-2, 1750, 81589, -473366}, // Vel Gam-2  1.750 8.158876 -47.336588
{Vel, Kap, 2470, 93686, -550107}, // Vel Kap  2.470 9.368562 -55.010668
{Vel, Lam, 2230, 91333, -434326}, // Vel Lam  2.230 9.133268 -43.432589
{Vel, Mu, 2690, 107795, -494203}, // Vel Mu  2.690 10.779488 -49.420255
{Vir, Alp, 980, 134199, -111613}, // Vir Alp Spica 0.980 13.419883 -11.161322
{Vir, Eps, 2850, 130363, 109592}, // Vir Eps Vindemiatrix 2.850 13.036278 10.959150
{Vir, Gam, 2740, 126943, -14494}, // Vir Gam Porrima 2.740 12.694345 -1.449375
```
For DSOs, I have another script that works and extracts Messier and Caldwell. The first argument is the catalog (M = Messier, C = Caldwell)

The fields are: 
- Messier number
- Constellation (see above)
- Object Type (also needs to be converted to a char)
- Magnitude (see above)
- RA (see above)
- Declination (see above)

```
./extract-dso.py M 20 

{001, Tau, Neb, 8400, 55755, 220144}, // M 1 Tau  Neb 8.4 5.57553 22.0144
{002, Aqr, GC, 6600, 215581, -8231}, // M 2 Aqr  GC 6.6 21.5581 -0.823056
{003, CVn, GC, 6300, 137031, 283756}, // M 3 CVn  GC 6.3 13.7031 28.3756
{004, Sco, GC, 5400, 163932, -265253}, // M 4 Sco  GC 5.4 16.3932 -26.5253
{005, Ser, GC, 5700, 153094, 20828}, // M 5 Ser  GC 5.7 15.3094 2.08278
{006, Sco, OC, 4200, 176724, -322542}, // M 6 Sco  OC 4.2 17.6724 -32.2542
{007, Sco, OC, 3300, 178975, -347928}, // M 7 Sco  OC 3.3 17.8975 -34.7928
{008, Sgr, Neb, 5800, 180614, -243803}, // M 8 Sgr  Neb 5.8 18.0614 -24.3803
...
{106, CVn, Gxy, 8700, 123160, 473042}, // M 106 CVn  Gxy 8.7 12.316 47.3042
{107, Oph, GC, 7800, 165422, -130528}, // M 107 Oph  GC 7.8 16.5422 -13.0528
{108, UMa, Gxy, 10200, 111920, 556733}, // M 108 UMa  Gxy 10.2 11.192 55.6733
{109, UMa, Gxy, 9900, 119600, 533744}, // M 109 UMa  Gxy 9.9 11.96 53.3744
{110, And, Gxy, 8200, 6728, 416906}, // M 110 And  Gxy 8.2 0.67275 41.6906

```
Works the same for Caldwell.

Need to extract NGC objects to replace the Herschel 400 and we are done.

The rest is to work is:

- Extract Herschel 400 objects
- Mapping the above mentioned values that need to be mapped to char ...etc., 
- Modify the code within SHC to use the catalog the same way they are used now
- Optionally add code for filtering, e.g. by magnitude

hjd1964, 2018-12-04:
Sure would like to have the star common names.

hjd1964, 2018-12-04:
Guess you're planning to map to long int's for RA/Dec...
Heck, might as well map to float if you're going to use 4 bytes.  It's sub-arc second with 7-8 digits of precision.

kbahey, 2018-12-04:
Nothing is cast in stone. As it stands, with long or float for coordinates, the whole struct is 12 bytes per item. So 220 stars + 110 each for Messier and Caldwell, and 400 Herschel comes down to 10,080 bytes, which is fine for Teensy 3.2 and ESP32.

Adding the proper name though would make them much bigger though ... We currently don't have them. That can be an enhancement in the future. 

For now, I just want the exact functionality that is in there to be still there, with more accurate coordinates. 

Anything else can be added in phase 2.

SlimDiesel, 2018-12-04:
> Sure would like to have the star common names.

Second that.

kbahey, 2018-12-04:
No problem. Can be in phase 2 as well ... I'd rather get more accurate catalogs with same functionality, than all the new fangled features to be done ...

kbahey, 2019-03-05:
This task has significant progress. 

The data is machine generated using Python programs that extract them from two sources: the Open NGC database, and the KStars list of stars.

There are now 408 stars, brighter than magnitude 3.85.

The Herschel objects are only 388 (missing 12, since I removed those marked Dup in Open NGC [duplicate?]).

Messier has 107 objects, which are all the ones marked as Messier in Open NGC.

Gotos are very precise now for all objects. A definite improvement over what was there before. 

If you want to test it, you can download it from my fork:
https://github.com/kbahey/OnStep/archive/shc.zip

Please see the other issue regarding improvements to the SHC here: #91

kbahey, 2019-03-05:
Pull request [107](https://github.com/hjd1964/OnStep/pull/107) submitted.

closed 2019-03-05 by kbahey