hjd1964/OnStep #100: Remove excess warnings

dlkeng, 2019-01-11:
Target: Classic_ON (Mega2560)
           MOUNT_TYPE_GEM
           No RTC
Arduino Preferences: Compiler warnings: All
 
In the latest OnStep releases, with all compiler warnings enabled, a bunch of warnings are generated. These are all associated with unused parameters, but should be corrected to clean up the code base.

```
In file included from HAL/HAL_Mega2560/HAL_Mega2560.h:56:0,
                 from HAL/HAL.h:27,
                 from OnStep.ino:61:
HAL/HAL_Mega2560/../drivers/RTCw.h:134:10: warning: unused parameter 'JD' [-Wunused-parameter]
     void set(double JD, double LMT) {

HAL/HAL_Mega2560/../drivers/RTCw.h:134:10: warning: unused parameter 'LMT' [-Wunused-parameter]

HAL/HAL_Mega2560/../drivers/RTCw.h:139:10: warning: unused parameter 'JD' [-Wunused-parameter]
     void get(double &JD, double &LMT) {

HAL/HAL_Mega2560/../drivers/RTCw.h:139:10: warning: unused parameter 'LMT' [-Wunused-parameter]

In file included from OnStep.ino:64:0:
lib/St4SerialMaster.h:151:6: warning: unused parameter 'baud' [-Wunused-parameter]
 void Mst4::begin(long baud)

AlignEq.ino:341:6: warning: unused parameter 'Lat' [-Wunused-parameter]
 void TGeoAlign::equToInstr(double Lat, double HA, double Dec, double * HA1, double * Dec1, int PierSide)

AlignEq.ino:406:6: warning: unused parameter 'Lat' [-Wunused-parameter]
 void TGeoAlign::instrToEqu(double Lat, double HA, double Dec, double * HA1, double * Dec1, int PierSide)

Home.ino:49:12: warning: unused parameter 'fast' [-Wunused-parameter]
 GotoErrors goHome(boolean fast)
```

dlkeng, 2019-01-25:
Updated the Arduino Core for ESP8266 to version 2.4.2: Now get these additional warnings - 6 warnings on line 254 and 6 warnings on line 337. 'mac' is an array of unsigned 8 bit values - they will always be >= 0 !!!
```
Wifi.ino: In function 'void processWifiGet()':

Wifi.ino:254:32: warning: comparison is always true due to limited range of data type [-Wtype-limits]
    if ((mac[0] >= 0) && (mac[1] >= 0) && (mac[2] >= 0) && (mac[3] >= 0) && (mac[4] >= 0) && (mac[5] >= 0))

Wifi.ino:337:32: warning: comparison is always true due to limited range of data type [-Wtype-limits]
    if ((mac[0] >= 0) && (mac[1] >= 0) && (mac[2] >= 0) && (mac[3] >= 0) && (mac[4] >= 0) && (mac[5] >= 0))
```

hjd1964, 2019-01-28:
Refactored to remove the Lat parameter from the equToInstr() and instrToEqu() routines.
Redesigned to correctly handle the mac address validation.
I will not be changing the design to avoid warnings about missing RTC parameters.

closed 2019-01-28 by hjd1964