hjd1964/OnStep #109: SHC: Data extraction/import issues

kbahey, 2019-03-06:
There are 12 objects missing from the Herschel catalog. Yet to be identified, and reported to the Open NGC project. 

For Messier, there are only 107 objects, instead of 110. The Open NGC data marks M102 as a duplicate. But it is also missing M45 (Pleiades) and M40 (Winnecke 4, a double star).

kbahey, 2019-03-07:
The Messier objects are now complete.

See [this commit](https://github.com/kbahey/OnStep/commit/4dda64c5fed5ca3370e188ba846d210386bf2eee).

hjd1964, 2019-04-15:
Oops, is this actually done?

kbahey, 2019-04-15:
Herschel contains 398 objects now. 

The details on why this is not 400 are in [this file](https://github.com/kbahey/etl-astro/blob/master/lookups.py#L197).

closed 2019-04-15 by hjd1964