hjd1964/OnStep #117: SHC development ideas

peterdulowsky, 2019-03-16:
I have few ideas of what could be improved in SHC:
1. Show active guide rate - perhaps top line at the left
2. Change "Value set" messages to meaningful messages like "Unparked" e.t.c.
3. Add option to Slew/Store Current position/Delete for custom catalogue items (stored in OnStep). 
4. Set Park, Return/Reset Home - ask for confirmation (to prevent accidental triggering of model clear)
5. Add symbols for Sidereal/Lunar/Solar tracking
6. Add last error clear - this will require more than SHC code change I believe
7. Goto when tracking disabled - Unknown error message - change to more meaningful message
8. Change Align page number to something like 10 - to allow for easier addition of custom screens (battery voltage in my case) - this is a very minor thing

kbahey, 2019-03-16:
Number 6 is already taken care of. Howard clears the error when tracking is enabled. 

Corollary to number 4: Add a way to avoid unintended presses of the 4 NSWE buttons. This is to protect for inadvertent guiding when exiting the menus. I find that I press the E button when exiting the menus. Perhaps add a 3 second sleep after exiting the menus before responding to these key presses? Or it can be a Guide menu from the main menu, and the 4 buttons will be enabled from that menu only?

Add a menu for guiding, and the 4 NSWE buttons will be enabled only when this

peterdulowsky, 2019-03-17:
Created a pull request that addresses 1, 2, 4 (partially) and 9

peterdulowsky, 2019-03-26:
Updated list:

~~1. Show active guide rate - perhaps top line at the left~~
2. Change "Value set" messages to meaningful messages like "Unparked" e.t.c. PARTIALLY DONE
3. Add option to Slew/Store Current position/Delete for custom catalogue items (stored in OnStep).  IN PROGRESS 
~~4. Set Park, Return/Reset Home - ask for confirmation (to prevent accidental triggering of model clear)~~
~~5. Add symbols for Sidereal/Lunar/Solar tracking~~
~~6. Add last error clear - this will require more than SHC code change I believe~~
7. Goto when tracking disabled - Unknown error message - change to more meaningful message
8. Change Align page number to something like 10 - to allow for easier addition of custom screens (battery voltage in my case) - this is a very minor thing

Edit 1:
~~9. Display OnStep & SHC firmware version~~

hjd1964, 2019-04-15:
Updated list 4/15/19:

~~1. Show active guide rate - perhaps top line at the left~~
2. Change "Value set" messages to meaningful messages like "Unparked" e.t.c. PARTIALLY DONE
3. Add option to Slew/Store Current position/Delete for custom catalogue items (stored in OnStep). IN PROGRESS
~~4. Set Park, Return/Reset Home - ask for confirmation (to prevent accidental triggering of model clear)~~
~~5. Add symbols for Sidereal/Lunar/Solar tracking~~
~~6. Add last error clear - this will require more than SHC code change I believe~~
~~7. Goto when tracking disabled - Unknown error message - change to more meaningful message~~
8. Change Align page number to something like 10 - to allow for easier addition of custom screens (battery voltage in my case) - this is a very minor thing

Edit 1:
~~9. Display OnStep & SHC firmware version7. Goto when tracking disabled - Unknown error message - change to more meaningful message~~

hjd1964, 2019-07-23:
Updated List 7/23/19:

> 2. Change "Value set" messages to meaningful messages like "Unparked" e.t.c. 

I added code for this a while back and will continue to keep the point in mind going forward.

> 3. Add option to Slew/Store Current position/Delete for custom catalogue items (stored in OnStep).

I don't see this happening for the foreseeable future.  If it does happen more likely it will be in the WiFi/Ethernet Add-ons (only.)

> 8. Change Align page number to something like 10 - to allow for easier addition of custom screens (battery voltage in my case) - this is a very minor thing

I'm not sure what this means, but the menu pages have been organized to make things easier to work on so I'm calling it done.

closed 2019-07-23 by hjd1964

peterdulowsky, 2019-07-24:
> > 1. Add option to Slew/Store Current position/Delete for custom catalogue items (stored in OnStep).
> 
> I don't see this happening for the foreseeable future. If it does happen more likely it will be in the WiFi/Ethernet Add-ons (only.)
That is exactly what I meant - it needs to be in the Wifi addon (only the catalogue data are in OnStep). I assumed a new page, Read and Write buttons and a text box that will show the User catalogue data in CSV format This way one could easily add, edit and delete own catalogues (used as observing lists). There is a thread somewhere on groups.io about it.