hjd1964/OnStep #142: Improve text on WiFi page

kbahey, 2019-06-28:
The current WiFi web page reads:

> "Enable either station OR AP mode, both enabled can cause performance issues
> However, if just setting up or testing it can be desirable to enable both modes temporarily to guard against being locked out.
> If locked out of the ESP8266, a Sketch uploaded to the MCU (Teensy3.x, STM32, Mega2560, etc.) which sends an 'R' at 9600 baud on the serial interface in reply to any '#' received will cause a reset to AP only enabled and the default SSID/Password."
> 

A better copy would be:

> "Enable either station <b>OR</b> AP mode, both enabled can cause performance issues."
> "However, if just setting up or testing it can be desirable to enable both modes temporarily to guard against being locked out.<br/>"
> "If locked out of the ESP8266, you can use the Arduino IDE to reset the flash memory (and hence go back to defaults.)  From the <b>Tools</b> menu, change <b>Erase Flash "Only Sketch"</b> to <b>Erase Flash "All Flash Contents"</b> and upload this WiFi-Bluetooth sketch again."

Also, the first two sentences can be improved to recommend enabling Station Mode and OnStep will fallback to its own AP mode.

peterdulowsky, 2019-06-29:
I agree the text can be improved. However, do not remove the option to have both AP and station mode on, please. 
Having both on works very well for me. When I have the mount set up at home it connects to the home wifi router. And when I am away I connect my phone to the mount via it's own AP. No need to change settings, no hassle, it simply works.
Thank you.

kbahey, 2019-06-29:
I did not mean to remove or change the functionality in any way. Only the text, to reflect the new "fallback to AP mode" which I use too.

It is just the text that can be improved. The second part of the text about recovery should be changed, to use the easier flashing method.

The first part of the text can be improved by rewording it somewhat to say that "if you intend to connect OnStep to your existing WiFi LAN, then use Station Mode, and uncheck AP mode. OnStep will fallback to AP mode if it cannot connect to your WiFi LAN, e.g. when you go to a dark site".

hjd1964, 2019-07-01:
How about this...

Enable either Station OR Access Point mode, both enabled can cause performance issues.  Note that even if Access Point mode is disabled OnStep will automatically re-enable it if the Station mode connection fails.  This can also be useful when you are away from home at a dark site and/or to guard against becoming locked out.

If locked out of WiFi you can use the Arduino IDE to reset the ESP8266 flash memory (and hence go back to defaults.)  From the Arduino Tools menu, change Erase Flash "Only Sketch" to Erase Flash "All Flash Contents" and upload the WiFi-Bluetooth sketch again.

kbahey, 2019-07-01:
Second part i sgood. 

First part may be:

Enable either Station OR Access Point mode, since enabling both can cause performance issues. If you want OnStep to connect to your home WiFi network, then enable only Station mode, and disable Access Point mode. If OnStep cannot connect to your home WiFi network, it will automatically re-enable Station mode. This is useful when you are away from home at a dark site and/or to guard against becoming locked out.

peterdulowsky, 2019-07-02:
Just to clarify, are the performance issues caused by 
a) having both AP and station enabled or 
b) by having both station and AP enabled AND simultaneously connected to a router/client? 
I have enabled both but do not use simultaneous connections and have seen no issues so far.

hjd1964, 2019-07-03:
Don't honestly remember, I think it was constant regardless of connections... but this was a long time ago so things might have changed in the libraries low level firmware.  It was a "stuttering" as if the ESP8266 was "slowly" switching between servicing each mode.

hjd1964, 2019-07-18:
These changes were made so I'm closing the issue.

closed 2019-07-18 by hjd1964