hjd1964/OnStep #144: add Big Screen 2.3" SSD1309 at SHC

dragonlost, 2019-09-08:
Hello.

I bought a 2.3 "screen based on SSD1309 I would like to know if it is possible to add it to the list of supported screens?

https://www.diymore.cc/collections/led-display-module/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow

ThomasW69, 2019-12-24:
I have the same display running. Here is how it works:
in config.h add:
`#define OLED_DISPLAY SSD1309    // SSD1309 is the 2.2"`

in SmartController.cpp add the bold text:
`if (model == OLED_SH1106)
    display = new U8G2_EXT_SH1106_128X64_NONAME_1_HW_I2C(U8G2_R0);
  else if (model == OLED_SSD1306)
    display = new U8G2_EXT_SSD1306_128X64_NONAME_F_HW_I2C(U8G2_R0);
  **else if (model == OLED_SSD1309)
    display = new U8G2_EXT_SSD1309_128X64_NONAME_F_HW_I2C(U8G2_R0);**`

in Smartcontroller.h add the bold text:
`#define SH1106 0
#define SSD1306 1
**#define SSD1309 2**`

and :
`public:
  enum OLED { OLED_SH1106, OLED_SSD1306**, OLED_SSD1309**  };`

in u8g2_ext.h add:
`//Intitialises big 2,3" display from diymore.cc
//Important: The display needs RESET pin. In my case it is Pin23 from EPS32 
class U8G2_EXT_SSD1309_128X64_NONAME_F_HW_I2C : public U8G2_EXT {
public: U8G2_EXT_SSD1309_128X64_NONAME_F_HW_I2C(const u8g2_cb_t *rotation, uint8_t reset = 23, uint8_t clock = SCL, uint8_t data = SDA) : U8G2_EXT() {
  u8g2_Setup_ssd1309_i2c_128x64_noname0_f(&u8g2, rotation, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
  u8x8_SetPin_HW_I2C(getU8x8(), reset, clock, data);
}
};`

hjd1964, 2019-12-31:
Preliminary support for the SSD1309 was added to the SHC, thanks Thomas.

closed 2019-12-31 by hjd1964