hjd1964/OnStep #146: [2.22] Platform Support Re-ARM LPC1768

Weruminger, 2019-10-10:
It would be nice to support the cheap and dowerfull Re-ARM Platform, that is a mainly 5V tollerant 32Bit arduino replacement. Nice would be some hints how to design a HAL to successfully build on Arduino IDE or on PlatformIO IDE. My current tryes ends up in a sellection of the DUE by PlatformIO Ide and not as expected the LPC1768 what works fine on my Marlin FW fork. but the HAL from Marlin, i do thik,, will not work on onstep caused by different define expectations.

if there is a How To create a HAL for OneStep, then i'd spend the time to realize and test it.

- HAL_ReArmRAMPS
- Pins.ReArmRAMPS.h
- Config.ReArmRAMPS.h  

Also i'd spend some basic config sets for EQ3, Exos-2 and EQ3 Thingiverse DIY Motor Set.
this may be a start for entry point OnStep installations for given mounts and motor Sets.

hjd1964, 2019-10-12:
There really is no HAL documentation other than what's in the comments.
I don't personally have an interest in supporting this, I will answer specific questions though to help you along.  Join OnStep at Groups.io

Probably the first thing you should do is make a pretend HAL from the HAL template and get OnStep to compile in the Arduino IDE with the LPC1768 as a target.  The most difficult task in designing a HAL is getting the hardware timers setup correctly:

Three timers are required, 32 bit are best.  If 16 bit with pre-scaler's that can work too but needs to be handled "just so."

One of the three is for the sidereal clock tick and timer supervisor, the other two are for step/dir outputs associated with Axis1 and Axis2.

All three must be setup to cleanly change their period without timing glitches or interruptions.  On some hardware like the ARM M4 Teensy's and ESP32 the timers do this without any effort.  On other hardware (Mega2560) the libraries sometimes don't handle things correctly and you need to directly set MCU registers, etc.

hjd1964, 2019-12-31:
This doesn't seem likely to happen so I'm closing it.

closed 2019-12-31 by hjd1964