hjd1964/OnStep #150: Better comment to avoid confusion

kbahey, 2020-02-08:
Jamie on the mail list was confused by this part, with a Gen-L and WeMos:

```
#define SERIAL_C_BAUD_DEFAULT         OFF //    OFF, n, ON for ESP32 Bluetooth.
#define SERIAL_C_BLUETOOTH_NAME  "OnStep" // "On..", Bluetooth device name for ESP32.
```

So better comments can help avoid this confusion:

```
#define SERIAL_C_BAUD_DEFAULT         OFF //    OFF, n, ON ESP32 internal Bluetooth. Only applies for the MaxESP PCB
#define SERIAL_C_BLUETOOTH_NAME  "OnStep" // "On..", Bluetooth device name for ESP32 MaxESP PCB.

```

hjd1964, 2020-02-10:
I just don't see the need for this nor is Bluetooth specifically a MaxESP feature (any ESP32) so I won't be changing the comment.  Perhaps a better idea would be to reconsider the name of the WiFi-Bluetooth Addon... since it doesn't have BT support.

closed 2020-02-10 by hjd1964

kbahey, 2020-02-10:
The confusion is real. See the linked message on how a new OnStep user interpreted: He saw bluetooth in OnStep's Config.h and assumed that there will be bluetooth. He did not notice the ESP32 part. 
Change it to any wording that would avoid this confusion.

I am not clear on what the Android App communicates with when using bluetooth? Does it send LX200 commands directly? So there is no web site when using bluetooth?
But yes, if the WiFi-Bluetooth is not needed when using an HC05, then yes, the module should be renamed to just WiFi.

hjd1964, 2020-02-10:
>
> The confusion is real. See the linked message on how a new OnStep user
> interpreted: He saw bluetooth in OnStep's Config.h and assumed that there
> will be bluetooth. He did not notice the ESP32 part. Change it to any
> wording that would avoid this confusion.
>

The ESP32 part is perfectly clear, his failure to read/understand caused
the problem.  Adding "MaxESP" implies you need a MaxESP2 board, that it
does something special, not the case.  And on top of that when you compile
it'll #error and tell you the feature isn't supported unless it's an ESP32.

I am not clear on what the Android App communicates with when using
> bluetooth? Does it send LX200 commands directly?
>
Correct, Bluetooth SPP protocol, acts like a serial port.

So there is no web site when using bluetooth?
>
Not unless the Wifi Addon is providing one.

But yes, if the WiFi-Bluetooth is not needed when using an HC05, then yes,
> the module should be renamed to just WiFi.
>
I figure when we rework the WiKi to match the OnStep release 3.16 we'll
address it.

kbahey, 2020-02-10:
OK, so it is clear that no addon is needed for bluetooth and it is just a direct serial port. That is good. Makes things simple, but no web site. That is not a limitation though.

As for renaming the WiFi-Bluetooth, it was my mistake then. I was the one that recommended that name, not knowing it is not needed at all. So, the repository needs to be renamed WiFi only, which also forces a WiFi-Bluetooth.ino to be renamed as WiFi.ino. This can be deferred to 4.x.

As for the comment in OnStep's Config.h, I thought that the ESP32 baud rate and bluetooth are only when one has a MaxESP board, so the builtin bluetooth feature of the ESP32 MCU is what is providing the connection.

So I am the one that is confused. Can you clarify this part to me?

hjd1964, 2020-02-10:
>
> As for renaming the WiFi-Bluetooth, it was my mistake then. I was the one
> that recommended that name, not knowing it is not needed at all. So, the
> repository needs to be renamed WiFi only, which also forces a
> WiFi-Bluetooth.ino to be renamed as WiFi.ino. This can be deferred to 4.x.
>

Actually at the time we were kicking around the possibility of supporting
ESP32 in addition to ESP8266 making BT a possibility.  At some point you
have to ask how many comms channels do you need and how many variations of
platforms and how many options.  Adds support issues for us and confusion
for users.

"SerialC" is Bluetooth on an ESP32.  On some other processors SerialC is a
> normal serial port or not available at all.  Checks are made during
> compilation.
>

The ESP32 SPP BT acts like any other Arduino serial port.  Serial.begin()
doesn't take an baud rate though, it takes the Bluetooth device name string
(OnStep) instead.  Since I can't check for strings to validate in the
pre-processor I use ON to enable Bluetooth (or a baud rate for normal
serial port, or OFF.)  Perhaps BLUETOOTH instead of ON would be nicer.