hjd1964/OnStep #167: Mega2560 Timer4 / PWM conflict

DavidObfuscator, 2020-09-21:
mega2560 boards configured to use timer4 as CTC (which is used for axis2)
meantime the timer4 handles the pwm functionality for the D6, D7 and D8.
D7/D8 are connected to mosfets on mks gen-l boards, which means, we cannot use 2 of the ports as pwm output

is it possible to change the timer4 for axis2? (maybe use timer0 for axis1 and use timer3 for the axis2 instead of axis1?)

hjd1964, 2020-09-22:
I really don't like the idea of introducing a change like this to the Mega2560 HAL.  Other users might have their own expectations of what's available and what's not as far as timers, pwm, and pins.

Then there is the question of the need for more fast pwm (aux feature ANALOG) since we already have two channels on heater outputs D9 and D10 (on Timer2.)  And dew heaters, if designed properly, are fine with slow pwm (software based aux feature DEW_HEATER) which can be on any pin.

DavidObfuscator, 2020-09-22:
If the timer4 will be dedicated for axis2, validation shall be extended to deny analog output for timer4 controlled pwm legs

closed 2020-09-26 by hjd1964