hjd1964/OnStep #186: maxRate bug

claudioarena, 2022-01-04:
I've noticed a problem with the maxRate variable and NV.
It's quite possible I've introduced the problem myself while modifying OnStep (apologies if so), but it seems to me that the problem is also present in the original codebase...

The problem:

When initializing the NV in initWriteNvValues() we call maxRateLowerLimit().
However, in maxRateLowerLimit() function, we have:
```
r_us=(r_us_axis1+r_us_axis2/timerRateRatio)/2.0;
```

The problem is the variable timerRateRatio. This is defined in Global.h:
```
volatile double timerRateRatio;
```

but not initialised. We only initialise it once we read the NV memory in initReadNvValues:
```
timerRateRatio    = axis1Settings.stepsPerMeasure/axis2Settings.stepsPerMeasure;
```

So when setting maxRate, we call maxRateLowerLimit, which is a problem because the non initialized variable initReadNvValues causes a division by zero.

I've fixed this in my version by simply moving the variable initialization to the definition:
```
volatile double timerRateRatio = axis1Settings.stepsPerMeasure/axis2Settings.stepsPerMeasure;
```

Am I missing something, or does the above make sense?

hjd1964, 2022-01-05:
Looks like a bug to me, if you're lucky enough to get past the first init. of NV all is ok as later it gets taken care of.

I didn't read your solution until I decided on the following change so I guess we agree roughly:

volatile double timerRateRatio = AXIS1_STEPS_PER_DEGREE/AXIS2_STEPS_PER_DEGREE;

... OnStepX code is **so** much easier to follow.

claudioarena, 2022-01-05:
Good to know it wasn't just me.

I've been a bit stuck on an RCX400 upgrade - servo motors and drivers not behaving super well, and PID tuning not tuned well yet.
So I will keep using OnStep for now, as that seems to have worked well so far.

Once I'm done, I'll definitely take a look at OnStepX - seems to be a really good upgrade to a good project to start with :)

hjd1964, 2022-01-05:
OnStepX is still being worked on, if running servos is important to you though maybe (or not) it is already worth looking into...

[https://onstep.groups.io/g/developer/topic/onstepx_and_servo_motors/87742678?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,87742678,previd=1641341884614851274,nextid=1640208076967172300&previd=1641341884614851274&nextid=1640208076967172300](url)

closed 2022-01-05 by hjd1964