hjd1964/OnStep #187: Meridian Flip Does not work anymore

azwing, 2022-01-25:
Howard,

may be I found a typo within OnStep.ino (a minus sign before degreesPastMeridianE)
Removing it Meridian Flip works again.
=====================================================================================

@@ -675,7 +675,7 @@ void loop2() {
         if (getInstrPierSide() == PierSideWest) {
           if (getInstrAxis1() > degreesPastMeridianW && (!(autoMeridianFlip && goToHere(true) == CE_NONE))) { generalError=ERR_MERIDIAN; stopSlewingAndTracking(SS_LIMIT_AXIS1_MAX); }
         } else
-        if (getInstrAxis1() < -degreesPastMeridianE) { generalError=ERR_MERIDIAN; stopSlewingAndTracking(SS_LIMIT_AXIS1_MIN); }
+        if (getInstrAxis1() < degreesPastMeridianE) { generalError=ERR_MERIDIAN; stopSlewingAndTracking(SS_LIMIT_AXIS1_MIN); }
       }
     }
     double a2; if (AXIS2_TANGENT_ARM == ON) { cli(); a2=posAxis2/axis2Settings.stepsPerMeasure; sei(); } else a2=getInstrAxis2();

azwing, 2022-01-26:
After many tests it seems the Meridian flip works after that change even I am not convinced the change is in the right place, it just prevents OnStep from stopping tracking issuing an underpole error.

azwing, 2022-01-26:
Afte some other test I discoverd I am stupid, but this is not new :-)

If all what I have understood from Limits seting is that if I set both degreesPastMeridianW and degreesPastMeridianE to the same value Autoflip will never happen.

But If I set degreesPastMeridianE to 4°
and           degreesPastMeridianW to -4° 
it works fine.

I am right?

closed 2022-01-26 by azwing

hjd1964, 2022-01-26:
The automatic meridian flip goto destination must be within limits (on the other pier side) or the goto will fail and it'll throw a meridian limit error and stop tracking.

azwing, 2022-01-27:
Thank you, 
I think there is a lot of confusion with this topic and would like to include a help in the Indi driver.