hjd1964/OnStep #188: Validate.h:380:4: error: #error "Configuration (Config.h): You must specify a PINMAP!"    #error "Configuration (Config.h): You must specify a PINMAP!"

tsatatwer, 2022-04-15:
Hi ,

while compiling I get this error:

**Validate.h:380:4: error: #error "Configuration (Config.h): You must specify a PINMAP!"
   #error "Configuration (Config.h): You must specify a PINMAP!**
   
   Any idea. Thanks

CyberianIce, 2022-04-15:
You must edit Config.h file according to your hardware. Error suggests you did not select PINMAP at all.

tsatatwer, 2022-04-15:
I did :

#define PINMAP                       AltMega2560 

Then I get many new errors ?????

CyberianIce, 2022-04-15:
For Arduino Mega 2560, You can use #define PINMAP Ramps14 and you can have all features on that board (no matter if you use ramps shield or not). And please join [OnStep group](https://onstep.groups.io/) to get community support, this is not proper place for that.

tsatatwer, 2022-04-16:
@CyberianIce many thanks indeed...already joined the OnStep group I appreciated the great support and very productive project .
I will close this issue

closed 2022-04-16 by tsatatwer