hjd1964/OnStep #193: Trojan:Script/Wacatac.B!ml (Trojan detected in OnStep 4.24 )

galvbass, 2022-12-10:
I just downloaded the last version of the OnStep and my windows Defender blocked a trojan in OnStep files: 
*OnStep-release-4.24 (2).zip->OnStep-release-4.24/addons/WiFi/Globals.h
*OnStep-release-4.24 (2).zip->OnStep-release-4.24/addons/Ethernet/Globals.h
*webfile:OnStep-release-4.24 (2).zip|https://codeload.github.com/hjd1964/OnStep/zip/refs/heads/release-4.24|pid:496,ProcessStart:133151572927477066
Please take action on this!

CyberianIce, 2022-12-10:
Perhaps the file is contaminated on your side. Please check your OS, browser or SSL proxy.

hjd1964, 2022-12-11:
False positive or as Cyberianice said is my interpretation, I see no other possibility.

closed 2022-12-11 by hjd1964

galvbass, 2022-12-11:
Thank you so much guys, everything is good was false positive.Thanks for all your support.Best regards.