hjd1964/OnStep #194: Error compiling for board DOIT ESP32 DEVKIT V1.

galvbass, 2022-12-25:
Hi, everyone Merry Christmas!
Please.....there is somebody able to save me from this Black hole!
Even if I verify in Arduino IDE the OnStep firmware without the boar or with board still showing this error.
I followed all the steps.
What I'm doing wrong?

OnStep- release 4.24
Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None, Disabled"

In file included from C:\Users\Alexandre\Documents\Arduino\OnStep\src\HAL\ESP32\ESP32.h:71,

                 from C:\Users\Alexandre\Documents\Arduino\OnStep\src\HAL\HAL.h:90,

                 from C:\Users\Alexandre\Documents\Arduino\OnStep\OnStep.ino:68:

C:\Users\Alexandre\Documents\Arduino\OnStep\src\HAL\ESP32\Analog.h:54:97: error: default argument given for parameter 3 of 'void tone(uint8_t, unsigned int, long unsigned int)' [-fpermissive]

 __attribute__ ((weak)) void tone(uint8_t pin, unsigned int frequency, unsigned long duration = 0)

                                                                                                 ^

In file included from sketch\OnStep.ino.cpp:1:

C:\Users\Alexandre\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:207:6: note: previous specification in 'void tone(uint8_t, unsigned int, long unsigned int)' here

 void tone(uint8_t _pin, unsigned int frequency, unsigned long duration = 0);

      ^~~~

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

CyberianIce, 2022-12-26:
Perhaps install ESP32 boards version 1.0.4 and "ESP 32 Dev Module" in the board selection

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

galvbass, 2022-12-27:
Thank you Cyberianlce! WORKING now with the version 1.0.4!

closed 2023-01-12 by hjd1964