hjd1964/OnStep #195: missing parameter in AlignEq ?

MichelMoriniaux, 2023-04-17:
I was reading through the formulas for creating the pointing model and all of them agree with Patrick Wallace's paper except:

AlignEq.ino line 155:
`TFd =TF*(cosLat*cosHa-sinLat*cosDec);`

https://sites.astro.caltech.edu/~srk/TP/Literature/Wallace_Telescope_Pointing.pdf page 6 defines TFd as:
`TFd = TF*(cosLat*cosHa*sinDec-sinLat*cosDec);`

is the omission of `sinDec` voluntary?

hjd1964, 2023-04-17:
Agree, the sinDec term should be there.

hjd1964, 2023-04-17:
I did some checks to be sure the math works with this change, and it looks good in a handful of orientations.

I won't be patching OnStep for this as it'd break alignment models, but I will make the change to OnStepX.

closed 2023-04-17 by hjd1964